flathub/org.freecad.FreeCAD #4: The Flatpak version of freecad doesn't allow you to open multiple freecad windows at the same time

K4LCIFER, 2024-10-30:
If I open a second Freecad window of the flatpak version, i just get a black window that doesn't respond; however, a regular freecad package will launch multiple windows just fine.

P.S. Please let me know if this sort of bug is better suited for an official Freecad bug repor rather than here, as I am unsure the proper place for this.

adrianinsaval, 2024-10-30:
possibly related code: https://github.com/FreeCAD/FreeCAD/blob/b6cda4309a9031bebffa73b694e9a9cc125d64de/src/Gui/Application.cpp#L2195

could be an upstream issue but AFAIK this has only been observed in flatpak version

adrianinsaval, 2024-10-30:
according to https://github.com/flathub/org.freecadweb.FreeCAD/issues/95 PID is always 2 in flatpak so that could explain it, the above function uses the PID for teh lockfile name

adrianinsaval, 2024-10-30:
confirmed, if I remove the lockfile the second instance runs without issues

adrianinsaval, 2024-11-06:
this has now been fixed upstream and will be available on flathub soon

closed 2024-11-06 by adrianinsaval