flathub/org.freecad.FreeCAD #15: White Screen in 3D View

Zindswini, 2024-11-16:
I installed the beta Flatpak (tag 1.0.0.39104) and the 3D View is a white screen.
This issue does not appear on the RC4 AppImage. Switching to software OpenGL does not resolve the issue.

FreeCAD Version Information:
```
OS: KDE Flatpak runtime (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 8364675)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

System Information:
```
Operating System: Aurora 41
KDE Plasma Version: 6.2.3
KDE Frameworks Version: 6.8.0
Qt Version: 6.8.0
Kernel Version: 6.11.7-300.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 7700X 8-Core Processor
Memory: 31.0 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3080/PCIe/SSE2
```

[image]

Zindswini, 2024-11-16:
Upon further investigation, this appears to be a Wayland + FreeCAD issue, not with the Flatpak. Disabling the Wayland socket in Flatseal "resolves" the error by forcing XWayland.

Related upstream issue:
https://github.com/FreeCAD/FreeCAD/issues/13893

luzpaz, 2026-09-30:
Should be fixed from 
* https://github.com/FreeCAD/FreeCAD/pull/23768

closed 2026-11-04 by adrianinsaval