flathub/org.freecad.FreeCAD #18: Not render CJK in Draft WB Text annotation

soiamsoNG, 2024-11-20:
on Fedora flatpak version can not render CJK in Draft WB Text annotation
same machine the appimage version render correctly.

[image]

OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

adrianinsaval, 2026-11-04:
this is likely the same issue as #23 and therefore should be fixed in beta branch, could you test?

adrianinsaval, 2026-11-09:
with the fix for #23 it still doesn't render with the default font but it works if I choose a font that explicitly has CJK characters, I imagine that is because my default system font doesn't have CJK characters. I see the same behaviour in the appimage so I'm inclined to believe this issue is now fixed in beta

soiamsoNG, 2026-11-09:
> with the fix for [#23](https://github.com/flathub/org.freecad.FreeCAD/issues/23) it still doesn't render with the default font but it works if I choose a font that explicitly has CJK characters, I [...]

I don't know freecad backend (if it follow th fc-math fc-list fallback system). but I have some experience on fontconfig
if your testing fontconfig  (Host 2.18.3,  Runtime 2.18.3 , use fc-math or fc-list to check if fallback order diffrent )
I think this is fontconfig issue, and flatpak (need install font.conf to fill the gap between host and runtime, like  https://github.com/flathub/com.tencent.WeChat/blob/master/com.tencent.WeChat.yaml  module install part ) it depend one your test locale,  LANG=en.UTF-8 ,  LANG=zh_cn.UTF-8 will have very different result if fc-match not using :codepoint query.

[65-lang.txt](https://github.com/user-attachments/files/34579786/65-lang.txt)