flathub/org.freecad.FreeCAD #19: BIM tutorial images are not vizible

ieugen, 2024-11-24:
Hi,

I'm following the BIM tutorial on FreeCAD 1.0 and I can't see the images.

This is what the report view shows me:
```
22:47:42  unparsable image path: /images/c/c4/BIM_Tutorial_37.jpg
22:47:42  unparsable image path: /images/5/55/BIM_Tutorial_38.jpg
22:47:42  unparsable image path: /images/4/42/BIM_Tutorial_39.jpg
22:47:42  unparsable image path: /images/7/73/BIM_Tutorial_34.jpg
22:47:42  unparsable image path: /images/5/52/BIM_Tutorial_40.jpg
22:47:42  unparsable image path: /images/thumb/b/b0/Draft_Dimension.png/16px-Draft_Dimension.png
22:47:42  unparsable image path: /images/1/1a/BIM_Tutorial_41.jpg
22:47:42  unparsable image path: /images/thumb/9/9f/Draft_Text.png/16px-Draft_Text.png
22:47:42  unparsable image path: /images/5/5e/BIM_Tutorial_42.jpg
```

luzpaz, 2024-12-02:
How did you access the BIM tutorial ?

ieugen, 2024-12-02:
From the Help menu, while on BIM Workbench
[image]

luzpaz, 2024-12-02:
This is not flatpak related because it's also a bug on the AppImage. Most likely it's related to the qtwebengine. I'll check upstream to see if there is a ticket open.

luzpaz, 2024-12-02:
It's fixed in the 1.1-dev weekly build (didn't make it into 1.0 release)  
* https://github.com/FreeCAD/FreeCAD/issues/17863

closed 2024-12-02 by ieugen