flathub/org.freecad.FreeCAD #23: Draft Text font selection does not work in Flatpak version

isloven, 2025-01-13:
If one creates a "Text" object in the Draft workbench in the Flatpak version of FreeCAD v1.0 (Revision 39109, Release Date 2024/11/18 16:48:00, Hash 2fcc531), changing the font using the the Data -> "Font Name" drop down box doesn't seem to have any effect (it seems to always use the default font).

Doing the same thing in the v1.0 AppImage version (Revision 39109, Release Date 2024/11/18 16:48:00, Hash 2fcc531) works correctly which means that is probably something with the Flatpak?

Few things that might help:
- This was tested on a clean install of Fedora 41 using default environment (Wayland, Gnome)
- The "Font Name" drop down box seems to correctly show all the installed fonts on the system
- There don't seem to be any error messages or warnings anywhere
- The documentation for the "Font Name" states that "If the given font is not found on the system, a default font is used instead." (src: https://wiki.freecad.org/Draft_Text) so it seems like it is not finding the select font - perhaps a path issue?
- Using Draft -> ShapeString and selecting the "Font File" path to the same font under /usr/share/fonts seems to work so it doesn't seem to be a flatpak permission issue

carlodri, 2026-10-05:
Same issue here on FreeCAD 1.1.4 (Flathub), Pop!_OS 24. I tracked down the cause and found a workaround.

**Cause:** Coin3D loads FreeType and fontconfig at runtime with `dlopen`, using the unversioned names (`libfreetype.so`, `libfontconfig.so`). Only the versioned `.so.N` files exist in the runtime, so the load fails. Coin then falls back to its built-in Times-like font and ignores `FontName` completely. Fontconfig inside the sandbox resolves the font fine (`fc-match` / `fc-list` work), so the fonts themselves aren't the problem.

**Diagnosis:**
```
$ flatpak run --env=COIN_DEBUG_FONTSUPPORT=1 org.freecad.FreeCAD
...
Coin info in cc_flw_initialize(): Win32 API can not be used for font support
Coin info in using_freetype(): FreeType library will not be used
```

**Workaround:** point Coin at the versioned libraries:
```
flatpak override --user \
  --env=COIN_FREETYPE2_LIBNAME=libfreetype.so.6 \
  --env=COIN_FONTCONFIG_LIBNAME=libfontconfig.so.1 \
  org.freecad.FreeCAD
```
After this, Draft texts and dimensions use the selected font.

**Suggested fix:** set these two variables in the manifest's `finish-args` (`--env=...`), or patch Coin's default library names at build time. This probably also fixes #18 (CJK not rendering in Draft text), since the built-in fallback font has no CJK glyphs.

Refs: [Coin environment variables](https://man.archlinux.org/man/coin_environment_variables.3.en), similar upstream reports [#2280](https://tracker.freecad.org/view.php?id=2280) and [#2704](https://tracker.freecad.org/view.php?id=2704).

hfiguiere, 2026-10-25:
Coin should be fixed to load DSO properly. Loading a bare `.so` for a versioned library is a bug, a fundamental misunderstanding on how DSO works on Linux (and other UNIX in General).

adrianinsaval, 2026-10-27:
@hfiguiere even if FREETYPE_RUNTIME_LINKING is disabled when building in coin the issue persists, I can see in the buildlog that cmake finds the unversioned .so:
```
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so (found version "2.14.3")
```
isn't it a bug in the runtime that the unversioned .so is in the sdk but not in the runtime?

hfiguiere, 2026-10-27:
>  isn't it a bug in the runtime that the unversioned .so is in the sdk but not in the runtime?

No. The .so is a symlink and could be pointing to an incompatible version. This is why DSO versioning exists. The .so are build time only to they are removed from the platform. This is not unique to flatpak runtime either.

adrianinsaval, 2026-10-30:
fixed in beta