flathub/org.freecad.FreeCAD #27: Middle click doesn't release

Jookia, 2025-02-20:
When using a mouse scheme like OpenInventor, clicking the middle button of the mouse to drag around causes the mouse to stay clicked unless I press mouse4. This happens in both stable and beta Flathub.

```
OS: KDE Flatpak runtime (i3/i3/xcb)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/86.4394
Installed mods: 
  * fasteners 0.5.31
```

Issue is not present in the Arch Linux build of FreeCAD.

adrianinsaval, 2026-11-05:
I can't reproduce on stable, please reopen if it's still an issue

closed 2026-11-05 by adrianinsaval

Jookia, 2026-11-06:
Oh yes, this issue has been fixed for a long time. :)